sir-buckyball/chrome-gcode-sender #35: Connecting MacBook Pro 2010 to X-Carve

millerama, 2015-08-13:
I am getting an error when trying to connect: 
Chrome did not use requested serial connection option. [ctsFlowControl; expected:false, actual:true]

Any advice on how to set ctsFlowControl to false?

Thanks,
Rick (noob)

sir-buckyball, 2015-08-14:
What version of chrome are you running? This is listed at chrome://chrome/

The bad news is that there is really not much that can be done in this case. The system is ignoring a parameter which may well be necessary for communication to be established properly. I know I have this issue on my 2006 mac mini, but have successfully controlled a shapeoko 2 (older version of an x-carve) from a 2009 macbook.

kevjett, 2016-04-01:
Any fix? I'm having same issue?

sir-buckyball, 2016-04-01:
As I said before, there is really not much I can do. It appears that there are some combinations of computers, operating system, and chrome versions where the Serial API of chrome does not do what is requested.

closed 2016-06-11 by sir-buckyball