sir-buckyball/chrome-gcode-sender #36: Looking for a successful x-carve run under windows 8

decarlowoodworks, 2015-08-28:
I have a x-carve by inventables and windows 8, I added gcode-sender to chrome. When I send the gcode, the first line, G20, is displayed in the console then I get a Alarm Lock and stops. I was able to get the Grbl version, 0.9j . I also toggled the connect/disconnect to see what would happen. It displayed $\x2=y .

Any help would be great!

sir-buckyball, 2015-08-29:
G20 tells your x-carve that you'll be sending coordinates in inches (as opposed to millimeters). It's a safe command in that your machine should do nothing upon receiving it. This means that your machine was likely already in an alarm lock state when you sent the first command.

There are multiple issues which can cause an alarm lock state:
- alarm pin is triggered (check that your limit switches are functioning properly with a multimeter)
- soft limits triggered (happens when the machine thinks that the carriage is outside the safety area; unlikely if you just turned on the machine)

It is possible to disable the alarms lock, but I strongly urge you to find the reason for the fault. This page discusses the matter more: https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

decarlowoodworks, 2015-08-29:
Thanks for the info. I did some more testing. I can't figure out why its failing on a G3, any ideas?

I exported Easel generated g-code and used the chrome gcode sender, it worked with out error. Then I used the Fusion 360 generated g-code. The only difference was the G3 command with I and J values and got an "error: Invalid gcode ID:33".

Here is the console:
error: Alarm lock
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]  ok
$X
[Caution: Unlocked]  ok
G20  ok
G90  ok
G1 Z0.15000 F9.0  ok
M9  ok
G0 X-0.0531 Y-3.3017 Z0.7225\xd Z0.3225  okokok
G1 Z0.1619 F30.0 Z0.1325\xd X-0.0529 Y-3.3019 Z0.1297\xd X-0.0525 Y-3.3028 Z0.1271\xd X-0.0519 Y-3.3041 Z0.1247\xd X-0.0511 Y-3.3059 Z0.1227\xd X-0.0501 Y-3.3081 Z0.1212\xd X-0.049 Y-3.3105 Z0.1203\xd X-0.0478 Y-3.313 Z0.12\xd X-0.0426 Y-3.3244
  okokokok
G17  okokokokokokok
G3 X-0.0261 Y-3.3305 I-0.0313 J-3.3192
error: Invalid gcode ID:33
M112
\x18
error: Unsupported command
Grbl 0.9j ['$' for help]

The g-code following the G3:

  G1 X0.0794 Y-3.2821
  X0.1818 Y-3.2324
  X0.2817 Y-3.1812

From M112 on is when I hit the E-STOP .

sir-buckyball, 2015-08-30:
From that console output, it looks as if your gcode file uses carriage-return characters (0x0D) instead of newline characters (0x0A).

As an experiment, would you mind switching which character you use to delineate the lines? I would guess that the line is being treated as a very long line, and is overrunning the buffer of your X-Carve, resulting in a truncated message.

If that experiment does the trick, we can update the file parsing routine (though I don't think 0x0D is an appropriate delineation character [it stems from a typewriter returning the carriage to the left]).

decarlowoodworks, 2015-08-30:
I tried it but got the same result. I'm assuming the I and J commands are valid. I made a similar design using Easel then exported the g-code. I noticed it did not generate I and J's to carve the arcs. Any ideas I can try?

I used Fusion 360 to generate the g-code, currently they do not have a post process for Inventables X-Carve. Do you have any recommendations?

sir-buckyball, 2015-08-30:
I don't suppose you'd mind sharing your problematic gcode file so I can play with it on my setup?

I have not used Fusion360. What output processors are available? I'd recommend picking the most generic sounding output.

FYI: I and J are vector coordinates to define the arc interpolation centerpoint (G3). Some programs just treat curves as a series of small lines which is why you may not see the G3 command in your Easel output.

decarlowoodworks, 2015-09-04:
I modified one of the post processors to generate R instead of IJK and was able to get it to run correctly. But I think IJK should work with X-Carve. Here's the gcode (w/IJK), maybe you can figure out why its not running. Thanks for the help!

G20
G90
G1 Z0.15000 F9.0
M3 S14000
G0 X-0.0531 Y-3.3017 Z0.7225 F9.0 
Z0.3225
G1 Z0.1619 F30.0Z0.1325
X-0.0529 Y-3.3019 Z0.1297
X-0.0525 Y-3.3028 Z0.1271
X-0.0519 Y-3.3041 Z0.1247
X-0.0511 Y-3.3059 Z0.1227
X-0.0501 Y-3.3081 Z0.1212
X-0.049 Y-3.3105 Z0.1203
X-0.0478 Y-3.313 Z0.12
X-0.0426 Y-3.3244
G17 
G3 X-0.0261 Y-3.3305 I-0.0313 J-3.3192G1 
X0.0794 Y-3.2821
X0.1818 Y-3.2324
X0.2817 Y-3.1812
X0.379 Y-3.1283
X0.4738 Y-3.0738
X0.566 Y-3.0178
X0.6557 Y-2.96
X0.6993 Y-2.9306
X0.7423 Y-2.9007
X0.7848 Y-2.8704
X0.8265 Y-2.8397
X0.8675 Y-2.8086
X0.9078 Y-2.777
X0.9474 Y-2.745
X0.9864 Y-2.7125
X1.0245 Y-2.6796
X1.062 Y-2.6463
X1.0987 Y-2.6125
X1.1347 Y-2.5782
X1.1699 Y-2.5435
X1.2044 Y-2.5084
X1.2382 Y-2.4727
X1.2711 Y-2.4366
X1.3033 Y-2.4
X1.3348 Y-2.3629
X1.3654 Y-2.3254
X1.3953 Y-2.2873
X1.4244 Y-2.2488
X1.4527 Y-2.2097
X1.4802 Y-2.1702
X1.507 Y-2.1301
X1.5329 Y-2.0896
X1.5581 Y-2.0485
X1.5825 Y-2.0069
X1.6061 Y-1.9647
X1.6289 Y-1.9221
X1.651 Y-1.879
X1.6722 Y-1.8353
X1.6927 Y-1.7911
X1.7125 Y-1.7464
X1.7315 Y-1.7011
X1.7497 Y-1.6554
X1.7672 Y-1.6091
X1.7839 Y-1.5623
X1.8 Y-1.5151
X1.8152 Y-1.4673
X1.8298 Y-1.419
X1.8437 Y-1.3702
X1.8568 Y-1.321
X1.8692 Y-1.2712
X1.881 Y-1.221
[... 511 lines cut ...]
X24.0079 Y-2.4435
X24.0438 Y-2.4777
X24.1179 Y-2.5456
X24.1946 Y-2.6116
X24.2739 Y-2.6761
X24.3556 Y-2.7391
X24.4399 Y-2.8006
X24.5267 Y-2.8606
X24.6158 Y-2.9192
X24.7073 Y-2.9763
X24.8011 Y-3.032
X24.8972 Y-3.0863
X24.9955 Y-3.1392
X25.0962 Y-3.1907
X25.1996 Y-3.2411
G3 X25.2163 Y-3.2353 I25.2051 J-3.2299
G1 X25.2218 Y-3.2241
X25.223 Y-3.2216 Z0.1203
X25.2241 Y-3.2192 Z0.1212
X25.2252 Y-3.2171 Z0.1227
X25.2261 Y-3.2153 Z0.1247
X25.2267 Y-3.214 Z0.1271
X25.2271 Y-3.2132 Z0.1297
X25.2273 Y-3.2129 Z0.1325
G0 Z0.7225M5G0 X0.00000 Y0.00000

sir-buckyball, 2015-09-08:
I've pushed a new release (2015.9.7.0). Please let me know if that does the trick for you.

This version adds support for the carriage return character being used as you have described it. This bug finally got me to add some tests to assert the expectations of the parsing code.

closed 2016-06-11 by sir-buckyball