sir-buckyball/chrome-gcode-sender #42: tiny gcode visualizer lib for tiny webserver

mkeyno, 2017-02-12:
sorry to open ticket for my question but actually I'm looking for light gcode visualizer or probably cad2gcode convertor in my webpage , but this webpage hosted on tiny ESP826612F module which is connected to Arduino GRBL gcode interpreter by serial line, so I should have great care about its weight , My plan is just connect to wifi module and open the browser and do designated CAM job.,As I'm not pro in client side script like JavaScript I was hopping could have good advise here, do you suggest to make some plugin from such library so I shouldn't have to download or call all library ?
thanks

sir-buckyball, 2017-02-12:
I'm not entirely sure what your questions are.

If you want to visualize some gcode and are using an ESP826612F as your server, then you probably don't want to do any rendering on it. Browsers are pretty powerful these days so you can just have javascript code on the page you are hosting do the rendering. If you're worried about the download size of the javascript library, consider using a CDN. github is capable of serving static content for you (like a javascript file).

The library that this chrome application uses for rendering also works in regular web pages. It depends upon the paperjs library. Here's another toy project which uses the same code. https://github.com/sir-buckyball/opencut/blob/master/src/gcode_render.js

mkeyno, 2017-02-12:
thanks @sir-buckyball  , my problem is most library which are visualizer  gcode files  or even  parse vector image to gcode has lot of dependency files which is out of my module power so I'm looking for light lib or  use some plugin so put most of this weight on browser side

sir-buckyball, 2017-02-13:
If you host the javascript files on a CDN and your users have a sufficiently modern browser, then it doesn't really matter how powerful your "server" is so long as it can serve a basic HTML file.

closed 2017-02-13 by sir-buckyball

mkeyno, 2017-02-13:
the module is in access point mode , so there is no internet connection