sir-buckyball/chrome-gcode-sender #47: Remote control

mglotzbach, 2017-12-25:
I am looking to use your software on a Chromebit device (https://www.asus.com/us/Chrome-Devices/Chromebit-CS10/).  I think it would be great if you could place the software in a remote mode, where a host computer could remotely control the app.  This would allow me to control the cnc from my development machine and send the file without that machine actually controlling the cnc itself.  It would naturally keep my development machine clean and dust free.

One idea about how it could work is to enable the device attached to the CNC in remote mode.  This would open a server socket.  A second computer, using the same software, could then connect to it, possibly using a simple discovery protocol.  Once connected the CNC host device would disable local controls and the remote client would have the same user interface available.  All the commands would be passed over a network, including the file to route.  It would also provide the added bonus of monitoring from remotely.  I could see a webcam feed in the future.

All this without additional cabling to control the cnc.  The system would run autonomous so it would not require the client to remain connected once the file is downloaded and started.

mglotzbach, 2017-12-25:
I forgot to add, I am very happy to do the work to add this feature.  I just wanted to see if would align with your vision of where this software could go.

sir-buckyball, 2017-12-26:
Interesting idea to use a chromebit in a remote mode. Personally I have a cheap chromebook in my workshop and share files with Google Drive to the account I use for it.

Would your chromebit have a monitor and keyboard connected to it or would it be completely headless?

My hesitation about this feature is that I think that octoprint could be used to similar effect. I believe it has considerably more features, would be cheaper than a chromebit, and has a larger community of members to support you with any issues. https://octopi.octoprint.org/

If you did add a remote feature to this codebase; I would expect it to use websockets. I don't know offhand what local discovery mechanisms are available for them (and am a little against needing a central server to facilitate such connections).