sir-buckyball/chrome-gcode-sender #53: Not splitting gcode at new line characters

12buckleo, 2019-06-24:
When trying to run Gcode exported from Fusion 360 (and possibly other CAM software) the program seems to send Gcode to the machine based on when a new G command is issued, rather than by newline characters

The splitting not working correctly results in either GRBL error 25 or 11 mostly since it is being sent many lines at once

Here is a sample of the Gcode that is not working for me, however there doesn't seem to be anything wrong with it so i'm unsure of how to fix the issue.

(1001)
G90 G94
G17
G21
G28 G91 Z0
G90

(2D Pocket1)
G54
G0 X104.324 Y81.735
Z15.2
Z5.2
G1 Z3.3 F500
X104.32 Y81.763 Z3.118 F333.3
X104.307 Y81.844 Z2.954
X104.283 Y81.969 Z2.822
X104.243 Y82.125 Z2.734
X104.189 Y82.297 Z2.7
G3 X98.811 Y80.407 Z2.387 I-2.689 J-0.945
X104.189 Y82.297 Z2.075 I2.689 J0.945
X98.811 Y80.407 Z1.762 I-2.689 J-0.945
X104.189 Y82.297 Z1.449 I2.689 J0.945
X98.811 Y80.407 Z1.137 I-2.689 J-0.945
X104.189 Y82.297 Z0.824 I2.689 J0.945
X98.811 Y80.407 Z0.511 I-2.689 J-0.945
X104.189 Y82.297 Z0.199 I2.689 J0.945
X98.811 Y80.407 Z-0.114 I-2.689 J-0.945
X104.189 Y82.297 Z-0.427 I2.689 J0.945
X98.811 Y80.407 Z-0.739 I-2.689 J-0.945
X104.189 Y82.297 Z-1.052 I2.689 J0.945
X98.811 Y80.407 Z-1.365 I-2.689 J-0.945
X104.189 Y82.297 Z-1.677 I2.689 J0.945
X101.5 Y84.202 Z-1.8 I-2.689 J-0.945

sir-buckyball, 2019-06-24:
So I've looked through the code and you are right that many of your commands are being sent as one large message. In general there should be a G command on each line, however I have seen before that some gcode generators will omit the command if they intend to only update the parameters of the previous command.

This bug smells very similar tohttps://github.com/sir-buckyball/chrome-gcode-sender/issues/38, which used Fusion360 to generate files like yours (with the actual command being omitted). In that case, only a single carriage-return character was used and the newline was omitted. There is handling for that case, but not if you have a carriage-return+newline sequence.

Could you attach the _exact_ file you are trying to process? In particular I am looking at the newline sequences used (which don't copy/paste very well).

I've also done some searching through the closed issues to see if there was a reason why newlines are being ignored, however nothing popped up. I _think_ this may have been purposeful to make sure that the rate limiting worked properly (eg, knowing when to wait for an OK versus not), however I cannot find proof of that in the code.

12buckleo, 2019-06-24:
Hey there, a couple of hours after posting that issue this morning I have discovered that fusion 360 can output gcode with a G command on every line and so I have infact used that to cut several things today and I can confirm that it is working.

I will attach the original file as a .nc anyway since it would be better if the default fusion post processor worked.

Link to gcode file: [File](https://drive.google.com/open?id=1KsJzEpTQ98s2QUbZd_tJf2x7Ak7CbruD)

The issue can be circumvented (for anyone else who comes across this issue in the future) by editing the post processor config. 
Search for gMotionModal and add **{force:true}** so that it matches as below:
var gMotionModal = createModal({force:true}, gFormat); // modal group 1 // G0-G3, ...

I have to say that once I got the program working, the gcode sender works brilliantly and even seems to continue when the screen turns to standby which is great. Good Job!