sir-buckyball/chrome-gcode-sender #55: codes not being recognized

gaetgodi, 2020-08-24:
unimplemented gcode command: M117
unimplemented gcode command: M300

sir-buckyball, 2020-08-24:
M series codes are machine specific.

If I had to guess, you are using Marlin firmware, where M117 corresponds to setting a message on an LCD panel, and M300 is for playing a tone. Neither of these codes affects the visualization of the code you are trying to send. If your machine responds normally, the presence should also not affect sending the file.

The warnings are intended to be rendered because it was as an unrecognized code, and without knowing your exact machine/firmware; it's safer to alert you to the issue.

Side note: Chrome is dropping support for chrome apps in a few months. I don't expect there will be any further updates to chrome-gcode-sender.

closed 2020-08-24 by sir-buckyball

gaetgodi, 2020-08-25:
Thank you, I am using an Ender 5 Pro, it is my first machine and I just got it. I like your gcode-sender, will it continue to work even if no longer supported?

sir-buckyball, 2020-08-25:
https://blog.chromium.org/2020/01/moving-forward-from-chrome-apps.html outlines the schedule when chrome apps will stop  working.

gcode-sender utilizes a special serial API, for which the alternative is not yet generally available.