sir-buckyball/chrome-gcode-sender #57: Google ends support for Chrome apps!

moof-moof, 2022-09-28:
I have been using gcode-sender, or rather my forked version of it, to control a homemade [2D jigsaw cutter](https://github.com/moof-moof/gee-saw-feeder). I find it a superb app, perfect for my use.
I recently became aware of the fact that Google is is soon migrating Chrome "Apps to Progressive Web Apps", and it worries me that gcode-sender may then stop functioning. I quote:

> ... the timeline for the phased approach for ending the support of Chrome Apps:
> 
> June 2022: Chrome Web Store will stop offering Chrome App installations for Windows, Mac, and Linux users.
> July 2022: End support for Chrome Apps on Windows, Mac, and Linux.
> January 2023: Chrome Apps will no longer be launchable on Windows, Mac, and Linux.
> January 2025: End support for Chrome Apps on ChromeOS

Do you have any advise on what to do avoid this threat, and hopefully continue using gcode-sender?

sir-buckyball, 2022-09-30:
yes... this has been kind of a long time coming.

The webSerial API is now available and should be usable as a replacement for the chrome.serial API which is used to actually connect to your machine and send bits. This API is also usable from regular web pages so the need for a chrome app isn't really a problem. https://developer.mozilla.org/en-US/docs/Web/API/Web_Serial_API

Offline use cases would need to use a progressive web app (PWA).

I've been meaning to make the necessary updates here, but time has been in short supply.

moof-moof, 2022-09-30:
Thanks for your quick response and explanation. 
I’m somewhat less worried now :-)

andrewhowden, 2023-09-13:
I came here from https://www.youtube.com/watch?v=lR5QjnohsRk. We (https://www.north27.co.uk/) support a number of Chrome OS based schools who I think would find this very useful. @sir-buckyball nice work! Maybe if time is short you could reach out for support here: https://summerofcode.withgoogle.com/

jgphilpott, 2024-03-04:
@sir-buckyball any advice on how exactly to use Web Serial API for sending Gcode to a printer? I've been trying for a while now but with no luck. I just posted [this question](https://3dprinting.stackexchange.com/questions/23119/can-i-use-web-serial-api-to-send-gcode-to-my-ender-5-pro) on the 3D Printing Stack Exchange to see if the internet community has any ideas.

**I'm not looking for a full app just a simple script that will connect and send a single Gcode to my printer... then I think I can take it from there.**

jgphilpott, 2024-03-04:
Awsome, I just found [the answer](https://3dprinting.stackexchange.com/a/23120/27036) to my own question! Turns out it is possible to send Gcode to a printer from a web browser with just a few lines of JavaScript! 🚀

jgphilpott, 2024-03-23:
@moof-moof @sir-buckyball @andrewhowden if anyone is interested I built [this simple terminal-style app](https://jgphilpott.github.io/polyslice/serial/browser/sender.html) for sending G-code to a 3D printer using [Web Serial API](https://developer.mozilla.org/en-US/docs/Web/API/Web_Serial_API). It should work with any [compatible browser](https://developer.mozilla.org/en-US/docs/Web/API/Web_Serial_API#browser_compatibility) (Chrome, Edge and Opera) and is also usable offline if you clone the repo and open the `sender.html` file in your browser.

moof-moof, 2024-03-23:
@jgphilpott 
Cool! I will have a look at your solution when I return from vacation.

andrewhowden, 2024-03-24:
Looks great! I'll hopefully have a chance to test over the Easter break.

moof-moof, 2024-04-23:
@jgphilpott 
I have now had time to test your terminal-style app. It looks very nice and well thought out. More importantly perhaps, using it it was super easy to establish a connection to the GRBL-based controller on my cnc-machine. Definitely a step-up compared to simply using the command line for control.

However, I have found a couple of "imperfections" that maybe you can help me to or improve.

Firstly, the return-stream from the device (GRBL in my case) does not display properly in the Output window. The lines are not printed out full-length (that is to the newline character) but instead chopped up randomly into very short lines, only about four characters long on average! This does not happen when using other terminal apps, so I guess the problem is not on the sending side. Please see attached screenshots showing the incorrect output versus what it looks like in a regular terminal. 

Do you have any ideas what may be the cause, and where in the code it may be fixed? 

Secondly, for my use-case it would be ideal to be able to upload complete gcode files from within the app via a file selection dialog, as opposed to just pasting the commands in chunks into the prompt textarea. 

Being more used to C and Python, but much less familiar with JavaScript, I could definitely do with informed ideas on how to implement that feature, too.

[image]
[image]

jgphilpott, 2024-04-24:
@moof-moof I'm glad you got the chance to test it out and that the connection process ran smoothly! With regard to your comments, I think it is best to move the conversation off of this thread. I have created two issues in my repo and added my response in the comments.

1) [Response data comes in small/short chunks making it hard to read](https://github.com/jgphilpott/polyslice/issues/1)
2) [Add the ability to upload a complete G-code file rather than typing or pasting commands into the text input](https://github.com/jgphilpott/polyslice/issues/2)

closed 2025-08-15 by moof-moof